clearwater/SwitecX25 #5: Port to Particle (particle.io) platform.

timkettering, 2015-09-20:
I had an existing X27 gauge project that originally used Ardiuno, which I wanted to replace with a Particle Photon.  

So I forked your library (https://github.com/timkettering/SwitecX25) and after changing the Arduino imports, the code compiled fine.  But the needle movement did not work as expected.  It took me the better part of the day but I tracked down the issue.  It seems that in `SwitecX25.h`, that the `dir` variable is defined as `char`.  This works fine on Ardiuno, but on Particle platform, when attempting to cast to int for comparison purposes, it does not work as expected.

Changing the type of `dir` to `int` resolved all issues.  I have committed changes and updated the wording for my fork and will maintain it as a Particle specific port of this library with all attribution to you as original developer.

Cheers!
-tim

timkettering, 2015-09-20:
Please do let me know if you have questions or any concerns.  :)

guyc, 2015-09-21:
Great to hear that you've ported to support Particle Photon, and it looks pretty straight forward.  I haven't had a chance to play with the Photon yet.  So many great platforms to choose from now.

Interesting issue you found... Just had a skim through the code looking at the use of dir, and it isn't obvious to me why dir shouldn't be a char type; it's not even used in any arithmetic operations.

timkettering, 2015-09-21:
I'm not certain why `char` is an issue.  I had to bring in a friend of mine who's more familiar with this lower level stuff than I am, and he says he suspects it is the cast of `-1` that is the issue.  

Here's a line of code he wrote to test his theory.

`char foo = -1; Serial.println((int)foo); // what is output?`

output: 255.

He thinks that it is architecture dependent.  On arduino, that may cast correctly to -1, but on Photon's ARM, it ends up as `255`.  

And in the X25 library, you are doing comparisons to determine whether the motor needs to move forward or back, and this seems to align with the issues I encountered at first..  needle would only move forward, but never back.

Cheers,
-tim

guyc, 2015-09-21:
Ahh of course, that's the root of it.  It appears that char is unsigned on the compiler for Particle Photon.  
It is signed on Arduino:  https://www.arduino.cc/en/Reference/Char

closed 2015-09-21 by timkettering