clearwater/SwitecX25 #9: problem running example

paulnd, 2017-01-02:
Hi,

I'm having a bit of a problem running the example on an eBay X27.168.

The motor runs to zero and then to about centre, so far so good. I type a value into the serial monitor but regardless of the value the motor moves to zero and then doesn't move at all following any further inputs into the serial monitor until I reset the board.

This happens on 2 different motors and I've tried swapping the motor connection to eliminate direction problems. The "needle" is a thin piece of tape so it's unlikely to be an acceleration problem and I've also connected the Arduino Mega to 12V.

I'll be grateful for any suggestions to get these motors working as this is a perfect solution for my gauge project.

regards

Paul

guyc, 2017-01-02:
Interesting, thanks for the report.  I haven't tested the serial example code for a long time.  Can you try printing nextPos just before ` motor1.setPosition(nextPos);` to determine if the problem is with the serial comms or with the motor driver itself?

paulnd, 2017-01-02:
Thanks for the quick response.

nextPos is 0 everytime so it is probably only the example sketch that has an error.

guyc, 2017-01-03:
Yep, sounds like the conversion from ascii to binary is failing for some reason.

Here's some sample code that doesn't depend on serial input - just randomly 
redirects the gauge.  It might give you some satisfaction that the motor
driver is working while you resolve the serial console issue.

```
#include "SwitecX25.h"

// 315 degrees of range = 315x3 steps = 945 steps
// declare motor1 with 945 steps on pins 4-7

SwitecX25 motor1(315*3, 4,5,6,7);

void setup(void) {
  digitalWrite(2,LOW);
  digitalWrite(3,HIGH);
  Serial.begin(9600);
  Serial.println("Go!");

  // run both motors against stops to re-zero
  motor1.zero();   // this is a slow, blocking operation
  motor1.setPosition(0);
}

void loop(void) {
  // update motors frequently to allow them to step
  motor1.update();

  if (motor1.currentStep == motor1.targetStep) {
    int position = random(0, motor1.steps);
    motor1.setPosition(position);
  }
}
```

paulnd, 2017-01-03:
Thanks, that works fine.

Now that I know the motor works I'll start working on my own sketch.

guyc, 2017-01-03:
FWIW I did fire up the sample app (OSX Arduino 1.6.5 with an Arduino Uno) and the serial code seems to be working for me.  Settings are baud rate 9600, end of line Line Feed.

[image]

closed 2017-01-03 by guyc