clearwater/SwitecX25 #12: Jittering on small movements

R5fan, 2017-09-27:
This is probably more a question than an issue with the code, but Im struggling to get smooth motion when the stepper is making small movements. Here is a demonstration:

https://www.youtube.com/watch?v=J-0FrQDX24s

When moving quickly from one fixed position to another, its all good. But in my application (airplane variometer) Im getting a stream of fluctuating values, sometimes big or so small that may correspond to just one or two stepper steps. The needle seems to move too quick from one step to another, causing this jittering?

I tried playing with the acceleration values, but I only managed to make it worse. Filtering the incoming data improves the result slightly , but at the expense of response time on large swings, so not really a solution either.

Im using adafruits x27.196 stepper connected directly to an arduino. This is my code;

`
#include "SwitecX25.h"
#include "Filters.h"

#define STEPS (315*3) // standard X25.168 range 315 degrees at 1/3 degree steps
SwitecX25 motor1(STEPS,2,3,4,5);

int target=0;
int ind1; 
int ind2;
int ind3;
int ind4;
int ind5;
String received;
String msgtype;
String variot;
float vario;
long previousMillis = 0;
long interval = 500;  
unsigned long currentMillis ;
float filterFrequency = 3  ;

// create a one pole (RC) lowpass filter
FilterOnePole lowpassFilter( LOWPASS, filterFrequency );  

void setup() {
 
  Serial.begin(19200);  
  vario=-0.1;
  // run the motor against the stops
  //motor1.zero();
  motor1.setPosition(STEPS);
  motor1.updateBlocking();   
  motor1.setPosition(565);
  motor1.updateBlocking();   
  
}

void loop()
{

  currentMillis = millis(); // test routing; move needle in 0.5m/s steps when no data is received on the com port
  if(currentMillis - previousMillis > interval) {
    previousMillis = currentMillis; 
    vario=vario+0.5;
     if (vario > 4.5) {      
      vario = -4.5;

    }
  }
  static char buffer[80]; // get data from openvario com port
  if (readline(Serial.read(), buffer, 80) > 0) {  
    received=buffer;
    previousMillis = currentMillis;  //reset test routine counter
    if (received.substring(0,4) == "$POV"){
 //     Serial.print("got vario:> "); 
        ind1 = received.indexOf(',');  //finds location of first ,
        ind2 = received.indexOf(',', ind1+1 );   //Position of P or E or V
        msgtype = received.substring(ind1+1, ind2);   //P or E or V
     // Serial.println(msgtype);
        if (msgtype == "E") { // vario message
          ind3 = received.indexOf(',', ind2+1 ); //value
          variot= received.substring(ind2+1, received.length()); //IAS
          ind4= variot.indexOf('*'); //position of checksum data at end
          variot= variot.substring(0,ind4); //strip checksumdata
          vario=variot.toFloat();          
        }
        
   }
    
  }
 
  target=lowpassFilter.input(-81*vario+565); //filtered
//  target=-81*vario+565; //unfiltered
  motor1.setPosition(target);
  motor1.update();
   
}

int readline(int readch, char *buffer, int len)
{
  static int pos = 0;
  int rpos;

  if (readch > 0) {
    switch (readch) {
      case '\n': // Ignore new-lines
        break;
      case '\r': // Return on CR
        rpos = pos;
        pos = 0;  // Reset position index ready for next time
        return rpos;
      default:
        if (pos < len-1) {
          buffer[pos++] = readch;
          buffer[pos] = 0;
        }
    }
  }
  // No end of line has been found, so return -1.
  return -1;
}`

Would using a stepper controller with microstepping be the solution, or am I missing something else?

guyc, 2017-10-05:
Thanks for the question - I especially appreciate the video, it really helps.  Let me describe what I think you are seeing, and you can tell me if I understand the problem correctly.

 - Your source is sending updates at some interval period (a few times per second by the look of it)
 - If successive positions are far apart, the needle moves smoothly between them.
 - If successives positions are not so far apart the needle takes less than that interval period to move between positions, so it moves quickly to the next position, then slows down or stops.
 - As a result, at certain speeds when you expect a nice slow needle sweep you are seeing a series of jerky movements.

I've seen a similar issue reading velocity from GPS units that send data at 1 second intervals, resulting in jerky motions each time an update is received.  This is something I'd like to solve, but it isn't straightforward.  Currently we are faithfully moving the needle to the next position as quickly as possible.  To give a smoother motion, we will have introduce some delay.  

The simplest solution would be to work out the update interval and slow down the transition so that it takes the full interval period to reach the new destination.  There are still some challenges with this because of the deceleration logic.  The downside is that you introduce latency - the needle will reflect the position a full interval period after the data was received.

It should be possible to reduce the latency by estimating what the next needle position will be based on the current angular speed and acceleration.  Doing that you will certainly overshoot from time to time - moving the needle to a position that lies outside of the actual values supplied by the source, which is an interesting side-effect.

It's an interesting problem and one that will probably take a few tries to get right.

Does this sound like your problem?

R5fan, 2017-10-05:
Hi Guy,

First of, thanks a lot for responding and all your hard work on this stepper!

I think the update frequency is only part of my problem. The update frequency of the values Im receiving is ~10-12Hz. However, when I replace the input with a potentiometer, Im getting similar stuttering. Here is my current prototype, this time controlled with a pot:

https://youtu.be/sdV95cUCXIQ

(In case you wonder why Im using 2 arduino's right now, its because the OLED has no framebuffer, and refreshing the screen using adafruit gfx library is blocking and takes just a weebit too long, causing the stepper to stutter, so Im using one arduino for the OLED, the other for the stepper. I'll replace both with an ESP32 eventually).

I think I'm seeing at least 2 things in that video: first, the limited resolution of the stepper, you can simply see it go from one step to the next. Secondly, its accelerating too fast between steps when the values vary only by a few steps, and because of that brisk acceleration, the needle "wiggles".  Its probably not helping that my needle is a little wobbly (kept it as light as possible), but well, it does help to show the problem :).

Ive meanwhile ordered a AX1201728SG. Im hoping microstepping will solve or at least hide the problem when using a pot meter. If so, I'll see what it does when using a 10Hz input signal and if maybe I need to interpolate. If not, Im not sure what I can do. 

BTW, are these steppers supposed to be so noisy?

R5fan, 2017-10-05:
Im uploading a better video:
https://youtu.be/9ljEoeU4Slw

R5fan, 2017-10-26:
I moved to a AX1201728SG stepper controller, and that solved the jitter. Closing this one.

closed 2017-10-26 by R5fan