clearwater/SwitecX25 #16: Motor looses steps

pingvincible, 2018-04-18:
Hi! I'm working on stepper control firmware with stm32 microcontroller. I wrote some code that can work with stepper and got a problem. I said to motor to move from position 1000 to 1111 and backward with speed of 50 steps every 20ms. And motor starts loosing its position with time. If I stops it and says to go to position 102(zero of my scale) it goes to the wrong place. I thought it happened because I didn't use acceleration and deceleration. Then I rewrite your code to use it in stm32 and had the same problem. Can you give me any advice in this case?

guyc, 2018-04-18:
In my experience, the cause of missed steps is usually:

 - too much acceleration (change the acceleration table)
 - too much torque (needle is too long or heavy)
 - low voltage (below 5v things get bad quickly)

To debug I suggest starting with a very slow non-accelerated forward-backward cycle and make sure that works reliably.  Use tape for the needle to eliminate needle torque problems.  If it still fails, look at the wiring and voltages.  Once that is confirmed working without losing steps, add a needle, and step up the speed.  I usually increase the speed until it starts to skip steps to give me some idea what the acceleration table should look like.

pingvincible, 2018-04-19:
Hi! Thank you for the feedback. After many hours of tests we have found our problem. Motor datasheet says that one must make 100ns delay between setting Dir and Step. And we didn't make it. Now everything works fine. Because your code works only with one motor and we control 8 motors at one time we decided not use your code in our project. Anyway thank you very much for your project it helped me a lot.

closed 2018-04-19 by pingvincible

nuonic, 2018-04-19:
Glad you found it.  For the record this library absolutely **does** work with multiple motors.

guyc, 2018-04-19:
Glad you found it. For the record this library absolutely **does** work with multiple motors.