clearwater/SwitecX25 #21: Wiring between arduino outputs and stepper motor pins

tinkeringbassplayer, 2023-03-07:
Hello Guy,

thank you for the provision of the stepper motor library. I was able to drive my X27.168 stepper motor with it but one question remains for me: How is the mapping between the four digital outputs of the microcontroller and the stepper motor intended ? I didn't find any mapping description.

From the image [http://guy.carpenter.id.au/gaugette/resources/2012-04-04/imgp9232.jpg](http://guy.carpenter.id.au/gaugette/resources/2012-04-04/imgp9232.jpg) you provided in another issue I can see that you use the output pins 4, 5, 6, and 7 of the microcontroller and the example code also uses the pins 4 to 7 for the initialization of the SwitecX25 object.

It is obvious that the white wire connects output 4 of the microcontroller to pin 1 of the stepper motor. The red wire connects output 7 to motor pin 4. But what about the remaining two black wires? From the picture I can't figure out which black wire goes to which pin as the black wires are twisted. I also swapped the connection of the black wires in the initialization: SwitecX25 motor1(STEPS,4,6,5,7) but I could not recognize any difference in the behaviour of the motor. So for me the question remains whether there is an intended wiring for the black cables or if it doesn't matter.

I would appreciate if you could replace the question marks in the following table with the intended pin number.

Arduino  Motor  remarks
Output   Input
=======================
4        1      the upper right motor pin in the picture (white wire)
?        2      the lower right motor pin in the picture 
?        3      the lower left motor pin in the picture 
7        4      the upper left motor pin in the picture (red wire)

regards
Marco

bjanders, 2023-03-15:
If you look at the source code then it assigns pin1, pin2, pin3, pin4. So in the example above I would assume Arduino pin 6 would go to motor pin 2 (lower right), and Arduino pin 5 would go to motor pin 3 (lower left).
I agree it would be very helpful if this was stated in the README.

guyc, 2023-03-16:
In fact the two black wires are interchangeable. 

https://guy.carpenter.id.au/gaugette/2012/04/04/making-wiring-harnesses/
[image]

bjanders, 2023-03-16:
Looking at the code and reading the datasheet, why do you even connect 4 wires instead of just 3, as pins 2 and 3 on the motor always have the same signal when not microstepping?

guyc, 2023-03-16:
I wouldn't recommend driving 2+3 from a single pin.  

The X25 spec sheet lists max current as 20 mA, and the ATmega328 microcontroller (Arduino UNO) has an absolute maximum rating of 40 mA source or sink per GPIO.  You can see in the step pattern that there are 2 states (1 and 4) where pins 1 and 4 are in opposition to 2+3, so a single pin would be subject to double current.  An inductive load induces flyback voltages, so the actual current may spike much higher.  It seems like an unnecessary risk.

bjanders, 2023-03-16:
You are right. I didn't think about that. A simple and cheap solution is to use a darlington array. Then you don't need to worry about the inductive load and current on the pins, and can save one pin on the microcontroller.

danonn1905, 2023-03-26:
доброго времени суток 
хотелось узнать возможности использования программы для А4988 и управления vid2905  в качестве спидометр,тахометр?

closed 2023-05-12 by guyc