clearwater/SwitecX25 #24: Setting pin mode in constructor does not work

goodlyduck, 2023-11-25:
Hi there

On my Arduino Nano ESP32, I had to also set pinMode to OUTPUT in my setup function, it does not seem to work to set them in the SwitecX25 constructor only. Perhaps add a setup() function that can do this later in runtime? I head that interfacing with hardware might be a bad idea in a constructor. 

/Martin

guyc, 2023-11-27:
Interesting, that's something I haven't run into before.  Any idea what the underlying cause is?

It sounds like pin modes are being reset after static initialisation?  does it work if you run the initialiser
within the setup loop, like this?

```
SwitecX25 *motor1;

void setup(void) {
  motor1 = new SwitecX25(315*3, 4, 5, 6, 7);
  ...
```

goodlyduck, 2023-11-29:
Thanks for your reply, I'm not sure what's causing this issue. It works if I do it like your proposal. The only difference then is that I have to use ->, i.e. motor1->zero() instead of motor1.zero().