clearwater/SwitecX25 #25: 3 pins vs 4 pins

hugomeiland, 2023-11-27:
According to X25 datasheet page 6; the bitmap for contact 2,3 are the same; in practice, I can disconnect gpio6 and connect contact2 & 3, and the motor will function as expected; Will it make sense to create 3-pin functions to allow to free the pin ;)?

guyc, 2023-11-27:
That's true, and has been discussed before:
https://github.com/clearwater/SwitecX25/issues/21#issuecomment-1472851814

I wouldn't recommend it for the reasons mentioned in that thread, i.e.:
> The X25 spec sheet lists max current as 20 mA, and the ATmega328 microcontroller (Arduino UNO) has an absolute maximum rating of 40 mA source or sink per GPIO. You can see in the step pattern that there are 2 states (1 and 4) where pins 1 and 4 are in opposition to 2+3, so a single pin would be subject to double current. An inductive load induces flyback voltages, so the actual current may spike much higher. It seems like an unnecessary risk.

guyc, 2023-11-28:
Closing - library will continue to require 4 pins for the reasons explained above.

closed 2023-11-28 by guyc

hugomeiland, 2023-11-28:
sounds fair, will it make sense to document this in the readme?
Thanks
Hugo