clearwater/SwitecX25 #27: Wrong StateMap

Saschko2000, 2023-12-22:
I've found out, that your Statemap:
// State  3 2 1 0   Value
// 0      1 0 0 1   0x9
// 1      0 0 0 1   0x1
// 2      0 1 1 1   0x7
// 3      0 1 1 0   0x6
// 4      1 1 1 0   0xE
// 5      1 0 0 0   0x8
static byte stateMap[] = {0x9, 0x1, 0x7, 0x6, 0xE, 0x8};

Is wrong for me. It is also wrong in the Datasheet. Maybe it is a difference for X25 and X27, but I don't think so.

[image]

If you follow the upper graph, it should be:
1: 1010
2: 1000
3: 0001
4: 0101
5: 0100
6: 0010

And not what the Bit Map below shows.

When I change your code to

static byte stateMap[] = {0b00001010,
                          0b00001000,
                          0b00000001,
                          0b00000101,
                          0b00000100,
                          0b00000010};

Everything runs perfectly :-)

guyc, 2023-12-23:
Interesting, you think the datasheet is wrong, and my code is wrong.
I assume your motor did not work with the original code?

This intrigued me so I had a careful look at the differences between the two bitmaps:

the original bitmap in my code, which matches the bitmap in the datasheet:
```
1: 1001
2: 0001
3: 0111
4: 0110
5: 1110
6: 1000
```

This works based on the following mappings as per the datasheet:

- contact 4: bit 3 (1000)
- contact 3: bit 2 (0100)
- contact 2: bit 1 (0010)
- contact 1: bit 0 (0001)
- ub1 is high when contact 1 > contact 2
- ub2 is high when contact 4 > contact 3

resulting voltages:
```
step 1 + +
step 2 + 0
step 3 0 -
step 4 - -
step 5 - 0
step 6 0 +
```

Your proposed bitmap is:
```
1: 1010
2: 1000
3: 0001
4: 0101
5: 0100
6: 0010
```
To get the correct final voltages, I believe you are working from the following assumptions.

- contact 1: bit 3 (1000) 
- contact 2: bit 2 (0100)
- contact 3: bit 1 (0010)
- contact 4: bit 0 (0001)
- ub1 is high when contact 1 > contact 2
- ub2 is high when contact 3 > contact 4

Note these differ from my assumptions in two significant ways: reversal of contact order and the polarity of ub2.
This would give us the same coil voltages, and match the pulse sequence in the data sheet
```
step 1 + +
step 2 + 0
step 3 0 -
step 4 - -
step 5 - 0
step 6 0 +
```

So my guess is that if you reverse the contacts on one of your coils (e.g. swap contacts 1 and 2) the original
bitmap would have worked.  Here's another way to look at it.

If we reverse the bit order on your bitmap:
```
1: 1010 -> 0101
2: 1000 -> 0001
3: 0001 -> 1000
4: 0101 -> 1010
5: 0100 -> 0010
6: 0010 -> 0100
```
Then reverse the polarity of the two high bits:
```
1: 1010 -> 0101 -> 1001
2: 1000 -> 0001 -> 0001
3: 0001 -> 1000 -> 0100 *vs 0111
4: 0101 -> 1010 -> 0110
5: 0100 -> 0010 -> 0010 *vs 1110
6: 0010 -> 0100 -> 1000
```
We end up with effectively the same as the original bitmap.  The values that vary only differ
in how they express a 0 voltage - low/low or high/high.

closed 2023-12-23 by guyc

Saschko2000, 2023-12-24:
Wow, you are right.
I like your way of thinking. 
And now I know, why they changed 00 to 11 (what is the same for a coil)
So the bits on 2 and 3 are the same, and we can drive the motor with only 3 Pins :-)
And Microstepping with PWM on Pin 1 and 4 should also work with 2 and 3 put together.
I hope I'll have some time in the near future to test it.

Thank you very much :-)