clearwater/SwitecX25 #30: Setting position too frequently causes controller to enter faulty state

crashkopf, 2024-02-08:
I wrote a quick Arduino sketch to try testing jitter, but encountered an interesting bug.  If position is set too frequently the controller takes smaller and smaller steps until its barely moving at all.  Everything works fine setting position every 500ms, but at 250ms intervals it breaks. 

[X25-Rando.zip](https://github.com/clearwater/SwitecX25/files/14215026/X25-Rando.zip)

I'll update this issue as I try to hunt down the root cause.

guyc, 2024-02-09:
Does the behaviour still happen if you remove the `Serial.println(position);`?

crashkopf, 2024-02-09:
It does.  I actually added that to make sure my position calculation wasn't just throwing out garbage.

crashkopf, 2024-02-09:
After instrumenting and playing with it a bit more, it really looks like at higher frequencies there's a damping effect that just causes the position to converge towards its average.  Apparently the PNRG library I'm using has a decently normal distribution.  The maximum velocity just acts like a low-pass filter.

The reason I'm playing around with this is my previous design had issues with the cheap Amazon steppers being noisy and exhibiting a lot of gear slop.  When using fast, real-time data that frequently jumped around in small increments it would get pretty annoying.  Of course, low-pass filtering helped a bit.  Now I've got some BKA30Ls to play with and wanted to test them to see if they were tighter and quieter.  So far, so good!

In trying to understand what was going on under the hood, I ended up refactoring the library a little bit.  I was able to simplify the update()/advance() logic and added a function to check if the stepper had come to a stop.  This actually turns out to be really useful, because I can choose whether or not to call setPosition() and I don't have to block while waiting.

If you're interested in the patch:
https://github.com/crashkopf/SwitecX25/commit/ffb95cc8e1db8ee35b5f35e93f75cbf44f1a934a

Closing this issue as it seems to be less of a bug and more of a general design limitation of the steppers.

closed 2024-02-09 by crashkopf

guyc, 2024-02-09:
Ahh yes, your analysis makes sense.

I appreciate the PR.  Given the large number of changes I can't merge it without extensive testing which isn't possible at the moment.

> added a function to check if the stepper had come to a stop

There is a property on the motor class `stopped` that is true if the motor is stopped.

crashkopf, 2024-02-09:
> Given the large number of changes I can't merge it without extensive testing which isn't possible at the moment.

I get that.  I  already found a race condition that intermittently stops everything when micros() overflows (takes 1.19 hours).  😆  I remember running into a similar situation trying to measure velocity from a rotary switch.  I was trying to read time between pin change interrupts, but if the next interrupt never came the 16-bit timer would just overflow.  I think the best solution would be to call update() from a timer interrupt, with time period as a parameter, but that has its own headaches.

>There is a property on the motor class stopped that is true if the motor is stopped.

I removed stopped along with dir, but then realized it needed the same check in two places, so I added it back as a function.  Maybe I just have a hard time understanding things without taking them apart and putting them back together. 😁 

I appreciate your taking the time to look at this.